nLabs-nScope/nScope-releases #1: Traces do not update when reading from 2 channels at 2mS/div

ndm736, 2016-01-08:
[image]

The traces do not update properly when reading from two channels at the 2mS/div time setting. Works fine reading from one, three or four channels, but not with two.

gruvin, 2016-02-03:
Confirmed same behaviour on Mac version also. (But I guess you knew that ;-) )

pfeerick, 2016-02-29:
Can also confirm. Works fine when reading one channel at 2mS/div, and spits out garbage and freezes when second channel is activated. only affects the 2mS/div setting - doesn't seem to affect any other time scales.

e-moe, 2016-03-18:
Confirmed on Mac OS X El Capitan

davidm2312, 2016-06-02:
Fixed in 4c0f2e62ae6ca3fa98631b4ba2897fde7c32d775

closed 2016-06-02 by davidm2312