nLabs-nScope/nScope-releases #2: Trace/Display does not update (in any situation) on my Windows 7 64 bit system.

gkientzy, 2016-01-20:
Trace/Display does not update (in any situation) on my Windows 7 64 bit system .   I know the unit does function as it works on my Windows 10 installation.  The display will update when power is turned off to display 'nScope is asleep (-_-)'.     That is the only update of the display.

davidm2312, 2016-01-20:
We haven't typically seen any issues with Windows 7 installations, so a few more details will help diagnose what might be happening.

Do Red/Blue LEDs on nScope light when you press the power button?
Does the power panel indicate that power is on and does it report the power usage properly?
Does the 'asleep' message disappear when nScope is not asleep?

Screenshots will be helpful, as well as detailed steps to reproduce the behavior.

gkientzy, 2016-01-21:
Here are more details and screenshots:
1) The LEDs work as expected - Green are ON and RED/BLUE turn on/off via power button.
2) It does show Power On at USB Power of .45W which seems correct.    Slightly changing over time so  it is making measurements.
3) The 'asleep' message DOES not disappear when nScope is not asleep.

Here is screen capture when apps starts up.
[image]

Here is screen capture when another app is dragged in front of nScope app - as you can see so refresh.

[image]

gkientzy, 2016-01-21:
The correct screen capture on startup is as follows - the previous capture was wrong:

[image]

gruvin, 2016-02-03:
Looks like it may be a video driver related issue. Do you have the latest drivers for your card / chipset installed? Or going further out on a limb -- because I'm only guessing it's even being used -- maybe something to do with your DirectX installation? Perhaps you could try reinstalling both the driver and DirectX, in that order? Worth a shot?

(Just a co-end-user offering 2 cents here.)