nLabs-nScope/nScope-releases #5: Not a Valid Application

frankgestrada, 2016-02-22:
Installed nScope on Windows XP version 2002 Service Pack 3 on a intel core i3 CPU, M370 2.40 GHz 909 MHz, 3.42 GB RAM.

Upon launch I get error "(null) is not a valid Win32 application"

pfeerick, 2016-02-29:
Unfortunately, that may be due to the dependencies and compile settings used to complete the executables for Windows. Programs that require the Microsoft dot NET framework will give that error message, as will some programs that have not been compiled to run in Windows XP. I'm not having any issues, but I am running Windows 10. I suspect XP will not be supported, unless it is an easy settings change. Windows 7 (or perhaps Vista) is likely to be the oldest version of Windows that will run the nScope software in its current form.

mpbiggs, 2016-03-06:
Ran nScope on my Win 10 desktop just fine, but also getting "nScope is not a valid Win32 application" when I try to run it on my Win XP laptop. Would be nice to have a solution for this, other than replacing or upgrading my laptop.

closed 2016-03-11 by davidm2312

mpbiggs, 2016-03-12:
> Closed #5 <https://github.com/nLabs-nScope/nScope/issues/5> via 7b55678

Thanks!

commit 7b556785 (2016-03-11): Added Linux RPMs and an updated windows MSI

Fixes #5