nLabs-nScope/nScope-releases #8: Request: input values by typing - sliders are not accurate enough.

marcelogaio, 2016-03-01:
The only issue (not a bug, but a request!) I am seeing so far (beside the 2ms/div bug) is that I am unable to select exact values (frequency, amplitude, Duty, time scale, etc). It would be very useful to be able to enter a value by typing (or something like that), because the sliders are not accurate enough.

For example, when trying to select Frequency or Duty Cicle, it's almost impossible to select a precise value with the slider. 

PS: I am using nScope on a MAC (OSX El Capitan)

saft6luck, 2016-03-01:
Just switched it on :-)
For windows version it's similar. I guess it's just not implemented in the beta right now ... but would be nice to get more precise ways to set the values.

davidm2312, 2016-03-03:
If you'd like some custom control, you can use the nScopeAPI to set these values manually, although at this point it's pretty undocumented.

Opening nScope GUI would overwrite whatever frequency values you set manually though.

marcelogaio, 2016-03-03:
Yes! I thought about the API... But even though It seems great for some particular programming, the main use of nScope (at least for me, and I guess many other people as well) is through the GUI.
Do you think this can be added? I guess making the fields editable would solve this issue.
Is the source code open?
Cheers!

davidm2312, 2016-03-03:
nScope isn't opensource as yet, but this would be a pretty nice feature. We'll see if we can implement it without cluttering up the interface.

saft6luck, 2016-03-03:
Great.
As you touch the GUI maybe you could add scales to the sliders and readings for displayed values.
Also, configuring the channels in tabs could be an option to gain space and make controls bigger and more precise.

The information about new firmware would be good, too. (Just selected the menu entry and the device went to "green flashing mode". Don't know if there was a new firmware but it started flashing and doesn't stop for 20 mins now.)

davidm2312, 2016-03-03:
Bigger controls are tricky with laptops with small screen resolution, there's always a tradeoff involved. 1200x700 is about as large as we can make the basic GUI.

You only need new firmware if your firmware and software versions mismatch. You can see this in the about menu. If you're continually flashing, then you should unplug and replug. It should be reporting this on the nScope GUI interface. Firmware updates only take about 30 seconds.

saft6luck, 2016-03-03:
Right, but how about bigger controls in combination with 1 tab per channel?

And very important (for me :-) ) to be able to shift the channels on the display ...

stevoclarke94, 2016-04-06:
I agree some kind of manual control for frequencies and such would be very useful... I found that once I have left 1Hz on the sig gen, actually getting it back to 1Hz is very difficult to do!

mpbiggs, 2016-07-30:
This is really a vital feature for the software. If I want to set a frequency or a duty cycle, I don't want "somewhere around 150Hz" or "something like 50%" (note that if you change the duty cycle from 50%, you can't get it back there again). I appreciate the progress you've made, but it is really important to be able to get exact settings.

ageeweb, 2016-08-07:
I need this to, need to set precise values

marcelogaio, 2017-04-15:
Did nScope project die? No more updates for a long time... :(

davidm2312, 2017-04-15:
Not dead, just dormant! We've suffered from a lack of build infrastructure, and for the past month or so we have been building up a build server to maintain nScope. More to come soon!

marcelogaio, 2017-10-02:
Any news on updates?

davidm2312, 2017-10-04:
yes, head to http://www.nscope.org/download/ for a preview

linux installers are still being brought online so no real activity on github yet.