nLabs-nScope/nScope-releases #16: nScope detected a power Fault ( O_O ) !!

tripoche, 2016-05-10:
Hello,
at startup of the software with nscope connected on the USB port I get the folowing message:
nScope detected a power Fault ( O_O ) !!
I get the same issue on 2 different platform (MAC and Windows)
Regards,
TR

KoliadaES, 2016-05-10:
The is most likely cause by plugging in to an non-powered hub or PC that can't deliver sufficient power.
USB is rated to 5V @ 500ma _per port_, but very few hosts/hubs can support this power rating.

tripoche, 2016-05-12:
Hello I have used USB ports direct connetion from the computer. I tested on all 4 USB ports (PC) and 2 USB ports (MAC Book). How can I investigate further ?
Thanks for your assistance.
TR

KoliadaES, 2016-05-12:
Assuming the LEDs on the nScope light (i.e. it is getting some power??) - find a powered hub that has sufficient rating and can deliver 500mA to at least one port (at a time). I use Etekcity from amazon but even that can only support 500mA in 3 ports concurrently (it is a 10 port hub). My desktop can also push 500mA. There is prolly a way to force a recalcitrant laptop to push the full 500mA, but I am not familiar.

tripoche, 2016-05-19:
1) LEDs on the nScope light
2) nscope software reports: " Power is Off: USB Power: 0.12W"
3) when I click on the up right button, Nscope reports "sleep mode"

I've tryed with 2 differents powerred USB hubs (Aukey 7 ports USB 3 and Trust 7 ports 14524). Still getting the same error message : nScope detected a power Fault ( O_O ) !!

Thanks for your assistance.
TR

davidm2312, 2016-05-22:
Sorry to hear about your problems. It's possible that you've received a board that has a hardware problem. To double-check everything, it's best to try nScope without the breadboard at all. Pull the board out of the breadboard, and try turning on the nScope using the button on the top. If the red and blue LEDs light, then we know that the breadboard has a defect. If you still experience the same problem, then we'll look to replace your nScope.

Let us know at contact@nscope.org

spicedreams, 2016-11-19:
Uh-oh, looks like I have this fault... First time I unboxed it, nScope connected to a PC's USB port, not via a hub, when nScope is on the breadboard I get this error, when disconnected I get red and blue lights OK.
Fortunately, I still have the breadboard from a previously-fault nScope and when I plug the new nScope into the old breadboard, I get red and blue lights OK.
I guess that means the breadboard supplied with the replacement nScope was faulty, but the previous one was good; so I have a working combination!

But; what can be wrong with a breadboard to cause this?

davidm2312, 2017-10-05:
I have seen some breadboards that are shorted internally. Always good to double-check!

closed 2017-10-05 by davidm2312