nLabs-nScope/nScope-releases #18: nscope dead

zackr6912, 2016-05-21:
The first time I plugged the nscope im it worked fine then I did a file\update firmware and the nscope disconnected from my PC and has been dead ever since. The 5v light flashes on when its plugged in but the nscope not recognized by my PC or any others i have plugged it into.

davidm2312, 2016-05-22:
Can you try plugging it in while the button is pressed? This should enter bootloader mode.  Then try again to file-> update firmware, and wait for the software interface to either tell you it's safe to unplug, or until it resets automatically

zackr6912, 2016-05-22:
it does the same thing. 5v light flashes and computer still not recognizing anything is plugged in.

davidm2312, 2016-05-22:
In bootloader mode, it may take a while to appear as a device to the computer, especially on windows. However, it should immediately turn on the right-hand green light and make it blink. If that doesn't happen, please try a different USB cable, we've actually been noticing some don't make a good connection.

If none of the above works to turn on any green light, then you may have somehow killed the nScope bootloader, which we'd like to know more about. Can you recall any messages that appeared on the screen when you clicked update firmware, or what was the process you took right as you did that action?

zackr6912, 2016-05-22:
Tried 2 other high quality USB cables. Still just a blue flash from the left side 5v light. When it happened it said "updating firmware...." then the nScope disconnected from my PC. The nScope software said "searching for firmware loader..." then "please unplug nScope and re-plug while holding button" which i did but the computer never recognized that I re-plugged it in.

davidm2312, 2016-05-23:
Send an e-mail to contact@nscope.org, and we'll get this sorted out.

closed 2016-05-23 by davidm2312