nLabs-nScope/nScope-releases #21: When using DC offset, channel does not update properly

KoliadaES, 2016-06-18:
Version 0.7

Start nscope
Connect channel 1 to +5V
Offset ch1 to -5V (or any negative voltage)

Trace does not update correctly.

Additional;
Now reduce the gain to 1V - trace disappears completely.

KoliadaES, 2016-06-18:
Actually, the 'additional' piece is not so - it appeared again...
Also, this seems somewhat random, I removed focus to write comment, went back to it and it was working properly.

davidm2312, 2016-06-18:
Ah, sure enough, thanks for the detailed description.

mpbiggs, 2016-07-30:
Easy to demonstrate the problem. Connect either pulse output to any channel. Offset the channel so any part of the pulse is below zero and the waveform falls apart (lots of little random triangles).