nLabs-nScope/nScope-releases #31: nScope binary requires libpng12, which is no longer supplied by Ubuntu

mgolden, 2016-11-21:
When I upgraded my (k)ubuntu system to 16.10, and re-installed nScope, it was failing to start up with this error:
$ nScope
nScope: error while loading shared libraries: libpng12.so.0: cannot open shared object file: No such file or directory

You can't install libpng12 from the ubuntu16.10 (Yakety) repos - because libpng 1.2 is quite old and really isn't supported anymore (see http://libpng.sourceforge.net/).

The workaround is pull the old version from old builds and manually install - see eg:
http://askubuntu.com/questions/838949/ubuntu-16-10-teamvieweri386-depends-on-libpng12-0

wget http://se.archive.ubuntu.com/ubuntu/pool/main/libp/libpng/libpng12-0_1.2.54-1ubuntu1_amd64.deb
sudo dpkg -i libpng12-0_1.2.54-1ubuntu1_amd64.deb

nScope should be rebuilt to use libpng 1.6, which is the most up-to-date version.

davidm2312, 2017-10-04:
Thanks for this.

To work around, add the following line to the software sources list
`deb http://cz.archive.ubuntu.com/ubuntu xenial main `

then install libpng12-0 with
`sudo apt-get install libpng12-0`

mgolden, 2017-10-04:
Do we know if this archive is making updates to libpng12 if there are security bugs? It's not a great idea to continue to depend on unsupported software.

davidm2312, 2017-10-04:
For v0.8.0, the toolchain is the default on Ubuntu 16.04 LTS, which is libpng12.

I’ll update the tool chain for the next release.

mgolden, 2017-10-05:
I notice that there is a site http://us.archive.ubuntu.com/ubuntu, and it seems to have libpng12-0, so US users could presumably use that.

05nelsonm, 2019-03-25:
When updates?