nLabs-nScope/nScope-releases #44: Green ligh flashing (GND) fault message overcurrent

eineltec, 2020-04-20:
hi guys

i have just plug the PCB and press the start button, and the green gnd lights start blinking.

a falut message appears in the console: nscope went overcurrent?

do you know wthat happends?

davidjmeyer, 2020-04-26:
That message means the nScope is consuming too much current, usually an indication of a short somewhere. Do you see this behavior when nScope is unplugged from any breadboard?

tlorach, 2023-01-30:
I never found the opportunity to use nScope, so I was excited to try it on a small Arduino setup... and after a minor manipulation, I ended-up in the same situation. I remember that I made a mistake of mixing ch1 with A1 and so plugging A1 to another source of current (a signal coming out from arduino) instead ot measuring it with Ch1. Since then, the device keeps complaining "overcurrent", plugged or unplugged from the breadboard.
I didn't even start to use this thing that it already broke ?? 
At least I am glad I didn't waste my time in learning how to use such a flaky thing that seems to be only good for the trash...